veebch/btcticker #112: Display always shows the running potato ;D "Issue: 0 Check your connection. Error at Line: 5102

TheRockaXXX, 2023-03-30:
Hi,
first of all, i really like your project. Thanks a lot.
But after my first (positiv) experience i always get the message above. I still can connect via ssh and the raspberry catches the right date and time. The connection to the network works, I think, the connection to coingecko failed. But I can't figure the reason.
Can someone help please?

veebch, 2023-03-30:
Running potato :)

Try logging onto the pi and pinging the coingecko api (curl https://api.coingecko.com/api/v3/ping) from there. 

If you can see it, try going back to the example config.yaml file, that will help you locate the problem.

TheRockaXXX, 2023-03-30:
Gecko says: "To the moon!" :D Ok, that works.
I checked the example config and compared it with mine (show below) and can't find any possible mistake.
`pi@cointicker:~/btcticker $ cat config.yaml 
display:
  24h: true
  cycle: false
  inverted: false
  locale: de_DE
  orientation: 90
  showrank: false
  showvolume: false
  trendingmode: false
ticker:
  currency: bitcoin
  exchange: default
  fiatcurrency: eur
  sparklinedays: 1
  updatefrequency: 120
`

veebch, 2023-03-30:
The whitespace in yaml files is important too. Try making a copy of your file and replading it with the example one. 

```
mv config.yaml config_old.yaml
cp config_example.yaml config.yaml
python3 btcticker.py
```

TheRockaXXX, 2023-03-30:
Same behavior. Same message on the screen :(. I think, I have to reinstall everything.  
I can't understand, why the connection to coingecko works (ping) and the price update doesnt. Maybe my ip is blacklisted?

TheRockaXXX, 2023-03-31:
I found out, that coingecko had a DDoS attack and was down a lot of time in the last few days. 
So, today i copied my modified config (same as before the problems) and it worked. 
Thanks for your fast and helpful support!!!

closed 2023-03-31 by TheRockaXXX