veebch/btcticker #123: Service running, no display

JBP3D, 2023-12-08:
Followed the steps, everything is running and had no errors but the display does not show anything

veebch, 2023-12-08:
There are some waveshare test codes in their repository. Can you display to the screen when you run them?

JBP3D, 2023-12-08:
> There are some waveshare test codes in their repository. Can you display to the screen when you run them?
> 
> EDIT: You might have V2 of the screen, if so see line 40 of the code. You need to comment out a line and then uncomment the one above

Thank you for the quick reply! My screen indeed has a sticker V2 on it so I'll try that and will update.

JBP3D, 2023-12-08:
> There are some waveshare test codes in their repository. Can you display to the screen when you run them?
> 
> EDIT: You might have V2 of the screen, if so see line 40 of the code. You need to comment out a line and then uncomment the one above

[image]

I did this but it's still not showing. Did I do it right?

veebch, 2023-12-08:
That looks good. What does output does the code give to the terminal when you run it?

JBP3D, 2023-12-10:
Hey sorry for the late response, had a few busy days.

I'm not sure what you mean with the output? I just used the commands to start the service and rebooted. using status shows btcticker is active running

[image]

veebch, 2023-12-14:
You can stop that service and then try running it interactively from the command line. 

The first thing I would do is try to get some waveshare test code running on it. That will help you narrow down the problem...

E-IG88, 2024-02-05:
I'm having the same issue. Was this ever resolved and if so, how? I'm running a Raspberry Pi 4B with a V2 screen.

E-IG88, 2024-02-05:
@veebch When you reference main.py, I'm assuming your talking about btcticker.py. I have already uncommented line 40 and commented out the one above it as shown below. I also searched the rest of the btcticker.py file for anything with the keyword "v2" and the only other section that has something starts at line 431 but I'm not sure what to do, if anything.

line 39: #from waveshare_epd import epd2in7
line 40: from waveshare_epd import epd2in7_V2 as epd2in7

def display_image(img):
    epd = epd2in7.EPD()
    # Also change to V2 if using a V2 screen
    epd.Init_4Gray()
    epd.display_4Gray(epd.getbuffer_4Gray(img))
    epd.sleep()
    thekeys = initkeys()
#   Have to remove and add key events to make them work again
    removekeyevent(thekeys)
    addkeyevent(thekeys)
    logging.info("Sent image to screen")
    return

veebch, 2024-02-05:
You're right.... btcticker.py

You shouldn't need to change the 431 bit. Can you get the waveshare demos to work?

E-IG88, 2024-02-05:
I've downloaded demos from Waveshare and can successfully display a "Hello World" demo to the screen so I know the e-ink display is working properly.

veebch, 2024-02-06:
Then it looks like gpiozero is doing it.

E-IG88, 2024-02-06:
@veebch This was it. I was running the script at the same time, or not closing Geany before running the code in the terminal. lgpio reserves that pin and won't let another process use it until it's been closed. I'm getting a lot closer now but still no display on the screen although it looks like it's trying to write to the screen. This is my output in terminal when running python3 btcticker.py --log DEBUG

username@raspberrypi2:~/btcticker $ python3 btcticker.py --log DEBUG
Traceback (most recent call last):
  File "/home/username/.local/bin/tzupdate", line 5, in <module>
    from tzupdate import main
ModuleNotFoundError: No module named 'tzupdate'
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'cycle': True, 'inverted': False, 'orientation': 180, 'trendingmode': False}, 'ticker': {'currency': 'bitcoin,ripple,ethereum', 'exchange': 'default', 'fiatcurrency': 'gbp,usd,btc', 'sparklinedays': 7, 'updatefrequency': 300}}
DEBUG:root:Setup GPIO keys
DEBUG:root:Add key events
INFO:root:Getting Data
DEBUG:root:[https://api.coingecko.com/api/v3/coins/bitcoin/market_chart/range?vs_currency=gbp&from=1706620854&to=1707225654](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapi.coingecko.com%2Fapi%2Fv3%2Fcoins%2Fbitcoin%2Fmarket_chart%2Frange%3Fvs_currency%3Dgbp%26from%3D1706620854%26to%3D1707225654&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C9432d3a5fc02473f745f08dc2716ee09%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428226645953503%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=wDwbCqCl2LvUy6ImoocZ20bTRS%2BkyUcLCJkCPinwvug%3D&reserved=0)
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): [api.coingecko.com:443](https://gcc02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fapi.coingecko.com%3A443%2F&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C9432d3a5fc02473f745f08dc2716ee09%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428226645961549%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=IRK96sRtsFdnsrGa9A3%2Busaqz8DsOHlv%2BiIPyyZyGL4%3D&reserved=0)
DEBUG:urllib3.connectionpool:[https://api.coingecko.com:443](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapi.coingecko.com%2F&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C9432d3a5fc02473f745f08dc2716ee09%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428226645968222%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=S%2Bn9hFCFqlXBxuJuGMJq20uCOoMdvNKBkgf13G%2Fin7E%3D&reserved=0) "GET /api/v3/coins/bitcoin/market_chart/range?vs_currency=gbp&from=1706620854&to=1707225654 HTTP/1.1" 429 187
DEBUG:root:Got price for the last 7 days from CoinGecko
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:Horizontal
ERROR:root:list index out of range
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:Horizontal
Traceback (most recent call last):
  File "/home/username/btcticker/btcticker.py", line 545, in fullupdate
    pricestack, ATH = getData(config, other)
                      ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/btcticker/btcticker.py", line 160, in getData
    timeseriesarray = rawtimeseries['prices']
                      ~~~~~~~~~~~~~^^^^^^^^^^
KeyError: 'prices'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/username/btcticker/btcticker.py", line 641, in main
    lastcoinfetch = fullupdate(config, lastcoinfetch)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/btcticker/btcticker.py", line 557, in fullupdate
    display_image(image)
  File "/home/username/btcticker/btcticker.py", line 435, in display_image
    epd.display_4Gray(epd.getbuffer_4Gray(img))
  File "/home/username/btcticker/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
    temp1 = image[i*2+j]
            ~~~~~^^^^^^^
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/username/btcticker/btcticker.py", line 663, in <module>
    main()
  File "/home/username/btcticker/btcticker.py", line 652, in main
    display_image(image)
  File "/home/username/btcticker/btcticker.py", line 435, in display_image
    epd.display_4Gray(epd.getbuffer_4Gray(img))
  File "/home/username/btcticker/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
    temp1 = image[i*2+j]
            ~~~~~^^^^^^^
IndexError: list index out of range

E-IG88, 2024-02-06:
Great, thanks! I will try it again later. Just out of curiosity, is it normal for all the other errors to occur if coingecko API refuses the connection, in particular "IndexError: list index out of range" ?

veebch, 2024-02-06:
Since the code was written, coingecko seem to be a lot more frequently throttling connections. That said, the code should spot the error code and stop gracefully, rather than try to run without data. I'll make that tweak next chance I get.

veebch, 2024-02-06:
The cup of tea rule has served me well in the past. I imagine you're hitting the url via your computer, which hasn't been throttled. If you get no error when accessing it via curl from the pi, that might be an issue

E-IG88, 2024-02-06:
Looks like your right. I have to time it almost perfectly but I finally got Coingecko to return an HTTP 200 OK. Still no display but I'm really close. Based from what I see near the bottom of the this traceback, it's still complaining about something related to the V2 screen. Does any of this make sense to you? Really appreciate all the help you've provided!

Traceback (most recent call last):
  File "/home/user/.local/bin/tzupdate", line 5, in <module>
    from tzupdate import main
ModuleNotFoundError: No module named 'tzupdate'
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'cycle': True, 'inverted': False, 'orientation': 180, 'trendingmode': False}, 'ticker': {'currency': 'bitcoin,ripple,ethereum', 'exchange': 'default', 'fiatcurrency': 'gbp,usd,btc', 'sparklinedays': 7, 'updatefrequency': 300}}
DEBUG:root:Setup GPIO keys
DEBUG:root:Add key events
INFO:root:Getting Data
DEBUG:root:[https://api.coingecko.com/api/v3/coins/bitcoin/market_chart/range?vs_currency=gbp&from=1706626799&to=1707231599](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapi.coingecko.com%2Fapi%2Fv3%2Fcoins%2Fbitcoin%2Fmarket_chart%2Frange%3Fvs_currency%3Dgbp%26from%3D1706626799%26to%3D1707231599&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984741736%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=a10PvqJ8oYmLRf97zaWyS7VXmCLdnwBPjGRF%2Ff%2F2eX4%3D&reserved=0)
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): [api.coingecko.com:443](https://gcc02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fapi.coingecko.com%3A443%2F&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984749986%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=gJqpHHv73VszywvClnhaR1KzbQDbpM6ITw0WH47RI30%3D&reserved=0)
DEBUG:urllib3.connectionpool:[https://api.coingecko.com:443](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapi.coingecko.com%2F&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984756323%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=5euOyUUMSA44I8PEQCreSDLfAAvFE%2FfLL6AdjamIGRc%3D&reserved=0) "GET /api/v3/coins/bitcoin/market_chart/range?vs_currency=gbp&from=1706626799&to=1707231599 HTTP/1.1" 200 None
DEBUG:root:Got price for the last 7 days from CoinGecko
DEBUG:root:[https://api.coingecko.com/api/v3/coins/markets?vs_currency=gbp&ids=bitcoin](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapi.coingecko.com%2Fapi%2Fv3%2Fcoins%2Fmarkets%3Fvs_currency%3Dgbp%26ids%3Dbitcoin&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984761600%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=WF8R7gidH%2BAZKtiCzCUIBMWOGip8uoWBcWG%2FWDNskG0%3D&reserved=0)
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): [api.coingecko.com:443](https://gcc02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fapi.coingecko.com%3A443%2F&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984766789%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=jvhW1isxwaVmpv2Gp1ESNGhHm531t96a4xhFeFNkMw4%3D&reserved=0)
DEBUG:urllib3.connectionpool:[https://api.coingecko.com:443](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapi.coingecko.com%2F&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984771882%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=WzGRJNyprcgouKGhulqR1BVoKNuSCRuyldcvEyDj66E%3D&reserved=0) "GET /api/v3/coins/markets?vs_currency=gbp&ids=bitcoin HTTP/1.1" 200 None
DEBUG:root:{'id': 'bitcoin', 'symbol': 'btc', 'name': 'Bitcoin', 'image': '[https://assets.coingecko.com/coins/images/1/large/bitcoin.png?1696501400](https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fassets.coingecko.com%2Fcoins%2Fimages%2F1%2Flarge%2Fbitcoin.png%3F1696501400&data=05%7C02%7Cevan.d.gatchek%40nasa.gov%7C3b267ee76c16443ab63908dc2724fa02%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638428286984776923%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=GZxjR%2BqQNJ2M1q%2FcnYdowNS3iN63V%2Fj49yzZ3c7QMdw%3D&reserved=0)', 'current_price': 34145, 'market_cap': 670352791362, 'market_cap_rank': 1, 'fully_diluted_valuation': 717500244958, 'total_volume': 13278747406, 'high_24h': 34335, 'low_24h': 33745, 'price_change_24h': -189.8194076939326, 'price_change_percentage_24h': -0.55284, 'market_cap_change_24h': -5894045018.616455, 'market_cap_change_percentage_24h': -0.87158, 'circulating_supply': 19620075.0, 'total_supply': 21000000.0, 'max_supply': 21000000.0, 'ath': 51032, 'ath_change_percentage': -33.00807, 'ath_date': '2021-11-10T14:24:11.849Z', 'atl': 43.9, 'atl_change_percentage': 77771.58417, 'atl_date': '2013-07-05T00:00:00.000Z', 'roi': None, 'last_updated': '2024-02-06T14:59:49.565Z'}
[... cut ...]
  File "/home/user/btcticker/btcticker.py", line 652, in main
    display_image(image)
  File "/home/user/btcticker/btcticker.py", line 435, in display_image
    epd.display_4Gray(epd.getbuffer_4Gray(img))
  File "/home/user/btcticker/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
    temp1 = image[i*2+j]
            ~~~~~^^^^^^^
IndexError: list index out of range
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner

E-IG88, 2024-02-06:
@veebch I finally got Coingecko to return an HTTP 200 OK. Still no display but I'm really close. Based from what I see near the bottom of my latest traceback, it's still complaining about something related to the V2 screen. Does any of this make sense to you? Really appreciate all the help you've provided!

DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'tEXt' 41 57
DEBUG:PIL.PngImagePlugin:STREAM b'pHYs' 110 9
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 131 3077
DEBUG:root:Getting token Image from Image directory
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:Vertical
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:Horizontal
ERROR:root:list index out of range
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy
DEBUG:waveshare_epd.epd2in7_V2:e-Paper busy release
DEBUG:waveshare_epd.epd2in7_V2:Horizontal
Traceback (most recent call last):
File "/home/user/btcticker/btcticker.py", line 550, in fullupdate
display_image(image)
File "/home/user/btcticker/btcticker.py", line 435, in display_image
epd.display_4Gray(epd.getbuffer_4Gray(img))
File "/home/user/btcticker/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
temp1 = image[i*2+j]
~~~~~^^^^^^^
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/user/btcticker/btcticker.py", line 641, in main
lastcoinfetch = fullupdate(config, lastcoinfetch)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/user/btcticker/btcticker.py", line 557, in fullupdate
display_image(image)
File "/home/user/btcticker/btcticker.py", line 435, in display_image
epd.display_4Gray(epd.getbuffer_4Gray(img))
File "/home/user/btcticker/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
temp1 = image[i*2+j]
~~~~~^^^^^^^
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/user/btcticker/btcticker.py", line 663, in
main()
File "/home/user/btcticker/btcticker.py", line 652, in main
display_image(image)
File "/home/user/btcticker/btcticker.py", line 435, in display_image
epd.display_4Gray(epd.getbuffer_4Gray(img))
File "/home/user/btcticker/waveshare_epd/epd2in7_V2.py", line 450, in display_4Gray
temp1 = image[i*2+j]
~~~~~^^^^^^^
IndexError: list index out of range
Exception in thread Thread-1:
Traceback (most recent call last):
File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner

veebch, 2024-02-06:
It's something in the waveshare code by the looks of it

E-IG88, 2024-02-07:
@veebch I got to the root of the issue. This was indeed an issue with the Waveshare code. 

In e-Paper\RaspberryPi_JetsonNano\python\lib\waveshare_epd\epd2in7_V2.py, change line 447 and 48000 in line 479 to 5808. The problem is thought to have been accidentally corrected by an engineer when adding a new screen. We will fix it in the next app update

https://github.com/waveshareteam/e-Paper/issues/322

veebch, 2024-02-08:
Excellent! Closing this issue as it seems to be a waveshare bug

closed 2024-02-08 by veebch