veebch/btcticker #124: Fresh OS, can't get it running...

ZOMGVTEK, 2023-12-17:
Why hello there.

On a fresh install, I cant manage to get this running. Pi3, 64bit lite (bullseye). Here is the output...

Traceback (most recent call last):
  File "/home/admin/.local/bin/tzupdate", line 5, in <module>
    from tzupdate import main
ModuleNotFoundError: No module named 'tzupdate'
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'cycle': True, 'cyclefiat': True, 'inverted': False, 'orientation': 90, 'trendingmode': False, 'showvolume': False, 'showrank': False, '24h': True, 'locale': 'en_US'}, 'ticker': {'currency': 'bitcoin,ethereum,cardano', 'exchange': 'default', 'fiatcurrency': 'usd,btc,gbp', 'sparklinedays': 1, 'updatefrequency': 300}}
INFO:root:Getting Data
ERROR:root:'FreeTypeFont' object has no attribute 'getsize'
Traceback (most recent call last):
  File "/home/admin/btcticker/./btcticker.py", line 548, in fullupdate
    image = updateDisplay(config, pricestack, other)
  File "/home/admin/btcticker/./btcticker.py", line 318, in updateDisplay
    tokenimage.thumbnail(resize, Image.ANTIALIAS)
AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/admin/btcticker/./btcticker.py", line 640, in main
    lastcoinfetch = fullupdate(config, lastcoinfetch)
  File "/home/admin/btcticker/./btcticker.py", line 554, in fullupdate
    image = beanaproblem(
  File "/home/admin/btcticker/./btcticker.py", line 247, in beanaproblem
    writewrappedlines(image, "Issue: "+message)
  File "/home/admin/btcticker/./btcticker.py", line 113, in writewrappedlines
    _place_text(img, line, 0, y_text, fontsize, fontstring)
  File "/home/admin/btcticker/./btcticker.py", line 102, in _place_text
    text_width, _ = font.getsize(text)
AttributeError: 'FreeTypeFont' object has no attribute 'getsize'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/admin/btcticker/./btcticker.py", line 662, in <module>
    main()
  File "/home/admin/btcticker/./btcticker.py", line 650, in main
    image = beanaproblem(str(e)+" Line: "+str(e.__traceback__.tb_lineno))
  File "/home/admin/btcticker/./btcticker.py", line 247, in beanaproblem
    writewrappedlines(image, "Issue: "+message)
  File "/home/admin/btcticker/./btcticker.py", line 113, in writewrappedlines
    _place_text(img, line, 0, y_text, fontsize, fontstring)
  File "/home/admin/btcticker/./btcticker.py", line 102, in _place_text
    text_width, _ = font.getsize(text)
AttributeError: 'FreeTypeFont' object has no attribute 'getsize'

TheWalrus057, 2023-12-18:
I have been experiencing the exact same same issue, have tried a few times to install but receive this series of errors. Researching it seems that Pillow has issues w the getsize function w the launch of 10.0.0, installing PIL 9.5.0 (pip install Pillow==9.5.0) got me through this error but it would crash upon cycling tokens citing "the LED is closed or uninitialized". With the autostart service created/enabled it runs again upon crashing but continues to show the first token set in config.yaml

Here is the traceback I get after installing PIL 9.5.0:
ERROR:root:LED is closed or uninitialized
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 96, in _write
    self.pin.state = self._value_to_state(value)
AttributeError: 'NoneType' object has no attribute 'state'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 580, in _check_open
    super(GPIODevice, self)._check_open()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 215, in _check_open
    raise DeviceClosed(
gpiozero.exc.DeviceClosed: LED is closed or uninitialized

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 549, in fullupdate
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 433, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 102, in module_init
    self.GPIO_PWR_PIN.on()
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 219, in on
    self._write(True)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 98, in _write
    self._check_open()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 583, in _check_open
    raise GPIODeviceClosed(str(e))
gpiozero.exc.GPIODeviceClosed: LED is closed or uninitialized

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 96, in _write
    self.pin.state = self._value_to_state(value)
AttributeError: 'NoneType' object has no attribute 'state'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 580, in _check_open
    super(GPIODevice, self)._check_open()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 215, in _check_open
    raise DeviceClosed(
gpiozero.exc.DeviceClosed: LED is closed or uninitialized

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 640, in main
    lastcoinfetch = fullupdate(config, lastcoinfetch)
  File "/home/pi/btcticker/btcticker.py", line 556, in fullupdate
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 433, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 102, in module_init
    self.GPIO_PWR_PIN.on()
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 219, in on
    self._write(True)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 98, in _write
    self._check_open()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 583, in _check_open
    raise GPIODeviceClosed(str(e))
gpiozero.exc.GPIODeviceClosed: LED is closed or uninitialized

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 96, in _write
    self.pin.state = self._value_to_state(value)
AttributeError: 'NoneType' object has no attribute 'state'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 580, in _check_open
    super(GPIODevice, self)._check_open()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 215, in _check_open
    raise DeviceClosed(
gpiozero.exc.DeviceClosed: LED is closed or uninitialized

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 662, in <module>
    main()
  File "/home/pi/btcticker/btcticker.py", line 651, in main
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 433, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 102, in module_init
    self.GPIO_PWR_PIN.on()
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 219, in on
    self._write(True)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 98, in _write
    self._check_open()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 583, in _check_open
    raise GPIODeviceClosed(str(e))
gpiozero.exc.GPIODeviceClosed: LED is closed or uninitialized

I love this project and appreciate everything you do veebch <3

veebch, 2023-12-19:
It looks like a combination of things. I think regressing Pil to 9.2 might do the trick, but it's better to change the code so that we can use the newest version.

I'll have a tinker with it later on and see if I can fix it.

closed 2023-12-19 by veebch