veebch/btcticker #128: Perhaps there are new package names/schemas

ScottTheHippy, 2024-02-20:
Howdy, 

I've attempted this with a clean install (username 'pi") and start having snags with the loading of dependencies.  Specifically here:

sudo apt-get update
sudo apt-get install -y python3-pip mc git libopenjp2-7
sudo apt-get install -y libatlas-base-dev python3-pil python-numpy python-matplotlib

The update is fine but all of these dependancies are failing.   Here are the messages:

Package python-matplotlib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-numpy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-numpy' has no installation candidate
E: Package 'python-matplotlib' has no installation candidate

Any ideas?  Thanks!

veebch, 2024-02-20:
maybe python3-numpy and python3-matplotlib?

ScottTheHippy, 2024-02-20:
I’m out of my depth on this, but I did see there was a ‘pip install numpy’ approach, but it didn’t get me any closer to working.  :(. If you want to have me try anything, I’ll gladly start with a fresh image and give more detailed feedback on the errors that come up.  Cheers!

veebch, 2024-02-20:
I'll leave this open until one of us figures this out :)

ScottTheHippy, 2024-02-21:
[image]

Are these warnings indicative of failed installations?

ScottTheHippy, 2024-02-21:
[image]
As a theory test, I tried just to pip install pandas after the prior error and it looks as though the installation succeeded.   I didn't mention prior, but in case it matters, this is a clean reimage on a RPi v3.

ScottTheHippy, 2024-02-21:
[image]
Then attempting to launch edify.py, this is the error (the same that I've been getting since I started).  
Cheers!

veebch, 2024-02-21:
edify.py?

ScottTheHippy, 2024-02-21:
Off the top- sorry for the confusion between btctracker and edify- I've been trying both and didn't figure that I'd make two issues over them seeming to have the same issue.  There was a clue in your reply.  I was using the "Raspberry Pi OS (Legacy 32-bit)" image instead of the "Raspberry Pi OS (Legacy 32-bit) Lite", which has Python and PIP (and Git) installed by default.  I just reimaged with Lite, installed PIP and Git, and followed the instructions posted on your edifier page and got similar errors (attached).  I'm about to do it again and follow the btctracker instructions and will post back when that's done.  
[image]

The first error message is during the install of the items listed in the requirements.txt.  Looks like it might be bumpy-specific.  
[image]

The second error message is what happened after I tried to execute the (edify.py) script.  

Thanks and more to come!

ScottTheHippy, 2024-02-22:
I was trying to figure out why my Waveshare demos print to the screen but btcticker doesn't and saw your comment in code about remarking one line and unremarking another if using "V2" of the display (which I am, based on the demos that work).  I followed your guidance and now get different error messages, and still no display.  Error attached.  Does that help anything?
[image]

ScottTheHippy, 2024-02-22:
Howdy Veebch,  I found your other thread about this V2 Waveshare business (https://github.com/veebch/btcticker/issues/123) and am happy to report I'm in business.  It threw me for a moment as I was editing the epd2in7_V2.py in the e-Paper\RaspberryPi_JetsonNano\python\lib\waveshare_epd directory and not seeing any headway.  I'd forgotten that your setup script copies those files to a waveshare directory with btcticker.  D'oh!  In any case, changing those lines was the last hurdle.  I don't know if you want to notate this  in addition to the other details (the Lite OS version, the "3" in the python install syntax) in the setup instructions or what, but YAY!  Thanks for your help!

closed 2024-02-22 by veebch