veebch/btcticker #131: ModuleNotFoundError: No module named 'urllib2'

gigac280, 2024-05-16:
May 16 11:14:38 piink systemd[1]: Started btcticker.service - btcticker.
May 16 11:14:41 piink python3[3160]: Traceback (most recent call last):
May 16 11:14:41 piink python3[3160]:   File "/home/admin/btcticker/btcticker.py", line 34, in <module>
May 16 11:14:41 piink python3[3160]:     import currency
May 16 11:14:41 piink python3[3160]:   File "/usr/local/lib/python3.11/dist-packages/currency.py", line 5, in <module>
May 16 11:14:41 piink python3[3160]:     import urllib2
May 16 11:14:41 piink python3[3160]: ModuleNotFoundError: No module named 'urllib2'
May 16 11:14:41 piink systemd[1]: btcticker.service: Main process exited, code=exited, status=1/FAILURE
May 16 11:14:41 piink systemd[1]: btcticker.service: Failed with result 'exit-code'.
May 16 11:14:41 piink systemd[1]: btcticker.service: Consumed 3.325s CPU time.

veebch, 2024-05-16:
Not a question, closing

closed 2024-05-16 by veebch

gigac280, 2024-05-16:
Bro... my question is error "ModuleNotFoundError: No module named 'urllib2' "

loiccoyle, 2024-05-20:
hi again @gigac280! you installed [`currency`](https://pypi.org/project/Currency) instead of [`nh-currency`](https://pypi.org/project/nh-currency/) somehow... you'll need to run `pip uninstall currency` and `pip install nh-currency`