veebch/btcticker #132: Waveshare V2 not working (NameError)

iddqd23, 2024-06-14:
If I comment out line 39 and uncomment line 41 in btcticker.py the screen stays white. If I additional change line 483 in btcticker.py to "epd = epd2in7_V2.EPD()" or "epd = epd2in7V2.EPD()" I get the error "NameError: name 'epd2in7_V2' is not defined". Any help?

veebch, 2024-06-14:
The second change (to line 438) is creating an error. Only the change of line 39 and 41 is needed. I'm not sure why the screen stays white, does it display the waveshare examples?

iddqd23, 2024-06-14:
No, the display stays white without any examples. Perhaps I have the wrong screen [DEBO EPA 2.7 Entwicklerboards](urlhttps://www.reichelt.de/entwicklerboards-display-epaper-2-7-schwarz-weiss-debo-epa-2-7-p224220.html?&trstct=pol_4&nbc=1).

When I start the btcticker.py via ssh, it displays the following message:

`sudo: /home/pi/.local/bin/tzupdate: command not found
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'24h': True, 'cycle': True, 'cyclefiat': True, 'inverted': False, 'locale': 'en_US', 'orientation': 270, 'showrank': False, 'showvolume': False, 'trendingmode': False}, 'ticker': {'currency': 'ethereum,cardano,bitcoin', 'exchange': 'default', 'fiatcurrency': 'usd,btc,gbp', 'sparklinedays': 1, 'updatefrequency': 300}}
INFO:root:Getting Data`

veebch, 2024-06-14:
That looks like the right board. The demo code on https://github.com/waveshareteam/e-Paper/tree/master/RaspberryPi_JetsonNano/python/examples should work on it. 

The output you posted is not saying anything about the code successfully getting the data, so it's possible coingecko isn't returning data

iddqd23, 2024-06-14:
The demo also don't work. I'm starting to get the feeling that my Pi Zero is broken. Although I can hardly imagine that, since access is possible via ssh.

veebch, 2024-06-14:
have you tried disconnecting the ePaper from the pi and then reattaching it? It *may* be that the screen isn't seated on the pins correctly

iddqd23, 2024-06-18:
I have now tried it with a new Raspberry Pi Zero and everything is working again. So it was actually my Pi that was broken. Many thanks for your help.

closed 2024-06-18 by iddqd23