veebch/btcticker #133: externally-managed-environment

will0913, 2024-07-01:
I was following along the instructions, and everything was working quite nicely until I hit
`python3 -m pip install -r requirements.txt`, where the error message says "This environment is extremely managed"

[image]

Therefore, I was getting errors about missing modules.

Your assistance is greatly appreciated.

veebch, 2024-07-01:
Here's a nice little article that should explain how to proceed:
https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi/overview

will0913, 2024-07-02:
Thanks for much for the quick response. Though other unrelated issues arose, the ones with venv have been resolved.

closed 2024-07-02 by will0913