veebch/btcticker #134: Failed to add edge detection

will0913, 2024-07-02:
After finally installed some of the missing modules, the latest issue seems to be with the display itself. 
I have a waveshare 2.9 inch V2, and I've changed some of the values in the btcticker.py. 
But I'm getting the following error message. How do I go about fixing it?

`sudo: /home/pi/.local/bin/tzupdate: command not found
INFO:root:epd2in9_V2 BTC Frame
INFO:root:{'display': {'cycle': True, 'cyclefiat': True, 'inverted': False, 'orientation': 90, 'trendingmode': False, 'showvolume': False, 'showrank': False, '24h': True, 'locale': 'en_US'}, 'ticker': {'currency': 'bitcoin,ethereum,cardano', 'exchange': 'default', 'fiatcurrency': 'usd,btc,gbp', 'sparklinedays': 1, 'updatefrequency': 300}}
ERROR:root:Failed to add edge detection
Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 660, in main
    addkeyevent(thekeys)
  File "/home/pi/btcticker/btcticker.py", line 515, in addkeyevent
    GPIO.add_event_detect(thekeys[0], GPIO.FALLING, callback=keypress, bouncetime=btime)
RuntimeError: Failed to add edge detection

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 716, in <module>
    main()
  File "/home/pi/btcticker/btcticker.py", line 705, in main
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 491, in display_image
    addkeyevent(thekeys)
  File "/home/pi/btcticker/btcticker.py", line 515, in addkeyevent
    GPIO.add_event_detect(thekeys[0], GPIO.FALLING, callback=keypress, bouncetime=btime)
RuntimeError: Failed to add edge detection`

veebch, 2024-07-02:
Does the 2.9 inch screen have buttons? If not then you might have to remove those parts of the code.

will0913, 2024-07-04:
Okay. I think I'm in over my head. I'll go through the code and see if I can figure it out. Thanks so much for the assistance.

closed 2024-07-04 by will0913