veebch/btcticker #135: Failed to start btcticker.service - btcticker.

Gustinos, 2024-07-17:
Hi,

Big noob here.
Am using v2 screen, already changed btcticker.py to the right one.
But im having two issues:

1:
when running python3 -m pip install -r requirements.txt i get the following:
[image]

2:
The service doesn't seem to run.
When checking systemctl status btcticker.service the following is sent:
[image]

Can you help me with these?

Thanks.

veebch, 2024-07-17:
1 causes 2, so you just need to fix 1 hopefully. 
The first bit is because you need to install the modules in a virtual environment.  Lots of distros have moved to this since the instructions were made. 

I'll update the instructions as soon as I get a chance.... but a quick google of 'python venv' should get you started if you get impatient :)

Gustinos, 2024-07-17:
I'm trying, no luck here.
Trying it with --break-system-packages, but get the following error:
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

any idea?

Gustinos, 2024-07-17:
Found that! Currently on failed to add edge detection

Gustinos, 2024-07-17:
Fixed by doing:
sudo apt remove python3-rpi.gpio

sudo apt install python3-rpi-lgpio

closed 2024-07-17 by Gustinos